mcuee/libusbk #4: libusbk does not list the USB device using WCID

mcuee, 2018-06-23:
Ref:
https://sourceforge.net/p/libusb-win32/mailman/message/36350250/

On Fri, Jun 22, 2018 at 4:33 PM, Chengda Guo <chengda.guo@...> wrote:
> I had used the libusbk and winUsb driver which created by the libusbk-inf-wizard.exe
> for installing my device, both of them are OK. And I can access the device by
> LibUsbk.dll in my application. Now I would like to automatically install winusb driver
> by Windows OS when the device plug into the system, and then use LibUsbk.dll
> to access the device in my application. The first step automatically install winusb
> driver was successful like below screen shot shows.
> But in my application, the device can't be accessed by LibUsbk.dll(Din't find
> the device, LstK_Init() succeed, but LstK_Count() return 0 device count).

mcuee, 2018-06-23:
Ref: 
https://github.com/pbatard/libwdi/wiki/WCID-Devices
https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/microsoft-defined-usb-descriptors

mcuee, 2018-06-29:
Apparently Microsoft's old recommendation on the WinUSB inf is using DeviceInterfaceGUIDs.
https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/winusb-installation#inf

```
;
;
; Installs WinUsb
;

[Version]
Signature = "$Windows NT$"
Class     = USBDevice
ClassGUID = {88BAE032-5A81-49f0-BC3D-A4FF138216D6}
Provider  = %ManufacturerName%
CatalogFile = WinUSBInstallation.cat
DriverVer=09/04/2012,13.54.20.543

; ========== Manufacturer/Models sections ===========

[Manufacturer]
%ManufacturerName% = Standard,NTamd64

[Standard.NTamd64]
%DeviceName% =USB_Install, USB\VID_0547&PID_1002

; ========== Class definition ===========

[ClassInstall32]
AddReg = ClassInstall_AddReg

[ClassInstall_AddReg]
HKR,,,,%ClassName%
HKR,,NoInstallClass,,1
HKR,,IconPath,%REG_MULTI_SZ%,"%systemroot%\system32\setupapi.dll,-20"
HKR,,LowerLogoVersion,,5.2

; =================== Installation ===================

[USB_Install]
Include = winusb.inf
Needs   = WINUSB.NT

[USB_Install.Services]
Include =winusb.inf
Needs   = WINUSB.NT.Services

[USB_Install.HW]
AddReg=Dev_AddReg

[Dev_AddReg]
HKR,,DeviceInterfaceGUIDs,0x10000,"{9f543223-cede-4fa3-b376-a25ce9a30e74}"

; [DestinationDirs]
; If your INF needs to copy files, you must not use the DefaultDestDir directive here.  
; You must explicitly reference all file-list-section names in this section.

; =================== Strings ===================

[Strings]
ManufacturerName=""
ClassName="Universal Serial Bus devices"
DeviceName="Fx2 Learning Kit Device"
REG_MULTI_SZ = 0x00010000

```

mcuee, 2018-06-29:
But then for WinUSB device mentioned here
https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/automatic-installation-of-winusb

Microsoft changed to use DeviceInterfaceGUID.

```

bPropertyData | 78 bytes | 7B 00 38 00 46 00 45 00 36 00 44 00 34 00 44 00 37 00 2D 00 34 00 39 00 00 44 00 2D 00 34 00 31 00 45 00 37 00 2D 00 39 00 34 00 38 00 36 00 2D 00 34 00 39 00 41 00 46 00 43 00 36 00 42 00 46 00 45 00 34 00 37 00 35 00 7D 00 00 00 | Property value is {8FE6D4D7-49DD-41E7-9486-49AFC6BFE475}.
-- | -- | -- | --

```

mcuee, 2018-06-29:
libusbK.dll needs to be changed to accommodate both DeviceInterfaceGUID and DeviceInterfaceGUIDs.

mcuee, 2018-06-29:
Work-around: use DeviceInterfaceGUIDs in FW. Tested under Windows 10 x64.

```
Microsoft Windows [Version 10.0.17134.112]
(c) 2018 Microsoft Corporation. All rights reserved.

C:\work>c:\libusbK-dev-kit\kList.exe

Loading USB ID's maintained by Stephen J. Gowdy <linux.usb.ids@gmail.com>..

 1. WinUsb Device (WinUsb Device) [Connected]
    Service              : WINUSB
    ClassGUID            : {88BAE032-5A81-49F0-BC3D-A4FF138216D6}
    DeviceID             : USB\VID_04D8&PID_0053\7&481749&0&2
    DeviceInterfaceGUID  : {9A75885D-0236-4E3B-A2A2-C6ECE85435F8}
    SymbolicLink         : \\?\usb#vid_04d8&pid_0053#7&481749&0&2#{9a75885d-0236-4e3b-a2a2-c6ece85435f8}
    DevicePath           : \\?\usb#vid_04d8&pid_0053#7&481749&0&2#{9a75885d-0236-4e3b-a2a2-c6ece85435f8}
    SerialNumber         : 7&481749&0&2
    BusNumber            : 0
    DeviceAddress        : 2

Select device (1-1) :1

Loading driver api..
Getting descriptors..

-Device:
  bLength             :18
  bDescriptorType     :0x01
  bcdUSB              :0x0200
  bDeviceClass        :0x00 (Defined at Interface level)
  bDeviceSubClass     :0x00
  bDeviceProtocol     :0x00
  bMaxPacketSize0     :8
  idVendor            :0x04D8 (Microchip Technology, Inc.)
  idProduct           :0x0053
  bcdDevice           :0x0000
  iManufacturer       :1 (Microchip Technology Inc.)
  iProduct            :2 (Microchip Vendor Example Device)
  iSerialNumber       :0
  bNumConfigurations  :1
!End Device
-Configuration:
  bLength             :9
  bDescriptorType     :0x02
  wTotalLength        :32
  bNumInterfaces      :1
  bConfigurationValue :0x01
  iConfiguration      :0
  bmAttributes        :0xC0
  MaxPower            :50 (100ma)
 -Interface:
    bLength             :9
    bDescriptorType     :0x04
    bInterfaceNumber    :0x00
    bAlternateSetting   :0x00
    bNumEndpoints       :2
    bInterfaceClass     :0xFF (Vendor Specific Class)
    bInterfaceSubClass  :0xFF (Vendor Specific Subclass)
    bInterfaceProtocol  :0xFF (Vendor Specific Protocol)
    iInterface          :0
   -Endpoint:
      bLength             :7
      bDescriptorType     :0x05
      bEndpointAddress    :0x01
      bmAttributes        :0x02 (Bulk)
      wMaxPacketSize      :64
      bInterval           :0x01
   !End Endpoint
   -Endpoint:
      bLength             :7
      bDescriptorType     :0x05
      bEndpointAddress    :0x81
      bmAttributes        :0x02 (Bulk)
      wMaxPacketSize      :64
      bInterval           :0x01
   !End Endpoint
 !End Interface
!End Configuration

C:\work>libusb\bin-1.0.22\examples\bin64\xusb.exe 04d8:0053
Using libusb v1.0.22.11312

Opening device 04D8:0053...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 0
           VID:PID: 04D8:0053
         bcdDevice: 0000
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: FF.FF.FF
       endpoint[0].address: 01
           max packet size: 0040
          polling interval: 01
       endpoint[1].address: 81
           max packet size: 0040
          polling interval: 01

Claiming interface 0...

Reading string descriptors:
   String (0x01): "Microchip Technology Inc."
   String (0x02): "Microchip Vendor Example Device"

Reading Extended Compat ID OS Feature Descriptor (wIndex = 0x0004):

  00000000  28 00 00 00 00 01 04 00 01 00 00 00 00 00 00 00  (...............
  00000010  00 01 57 49 4e 55 53 42 00 00 00 00 00 00 00 00  ..WINUSB........
  00000020  00 00 00 00 00 00 00 00                          ........

Reading Extended Properties OS Feature Descriptor (wIndex = 0x0005):

  00000000  90 00 00 00 00 01 05 00 01 00 84 00 00 00 01 00  ................
  00000010  00 00 2a 00 44 00 65 00 76 00 69 00 63 00 65 00  ..*.D.e.v.i.c.e.
  00000020  49 00 6e 00 74 00 65 00 72 00 66 00 61 00 63 00  I.n.t.e.r.f.a.c.
  00000030  65 00 47 00 55 00 49 00 44 00 73 00 00 00 50 00  e.G.U.I.D.s...P.
  00000040  00 00 7b 00 35 00 38 00 64 00 30 00 37 00 32 00  ..{.5.8.d.0.7.2.
  00000050  31 00 30 00 2d 00 32 00 37 00 63 00 31 00 2d 00  1.0.-.2.7.c.1.-.
  00000060  31 00 31 00 64 00 64 00 2d 00 62 00 64 00 30 00  1.1.d.d.-.b.d.0.
  00000070  62 00 2d 00 30 00 38 00 30 00 30 00 32 00 30 00  b.-.0.8.0.0.2.0.
  00000080  30 00 63 00 39 00 61 00 36 00 36 00 7d 00 00 00  0.c.9.a.6.6.}...

Releasing interface 0...
Closing device...

```

mcuee, 2018-07-01:
Ref: https://www.osronline.com/showthread.cfm?link=289556

TravisRo, 2021-06-13:
Yes. Microsoft do that to us.  I've added support for both "DeviceInterfaceGUID" and "DeviceInterfaceGUIDs" to the DLL but the libusbK driver will still require the latter.

closed 2021-06-13 by TravisRo