mcuee/libusbk #12: To release 3.10.0

mcuee, 2018-06-30:
This is used to track the milestone.

TravisRo, 2021-06-13:
We can do a release but I do not have a valid certificate to sign the driver.

mcuee, 2021-06-13:
@TravisRo Great that you are back in action. I think it is okay to release an unsigned version first.

mcuee, 2021-06-13:
To me libusbk driver is not that important any more , rather the DLL is still good to carry forward. Therefore the unsigned driver is not a big issue.

mcuee, 2021-06-15:
@TravisRo 
My idea is like the following.
https://github.com/mcuee/libusbk/milestones.
1) Release what we have as 3.0.8 soon.
2) If you got the time to implement WinUSB isoc support, then we can set that as a goal for release 3.0.9.

mcuee, 2021-07-09:
3.0.8 has been released.

mcuee, 2021-07-20:
3.1.0 has been release with major new features.
```
V3.1.0.0 (07/19/2021)
==============================================
- Added isochronous support for WinUSB

- Added BOS descriptors

- Added MS OS V1.0 Descriptors

- Added MS OS V2.0 Descriptors

- Added 'DeviceInterfaceGUIDs' property to libusb0 .inf template

- Fixed issues causing missing devices in the device list when then
  DeviceInterfaceGUIDs property is missing
  
- Added BOS and MS descriptor reporting to kList

- Added FX3 benchmark test firmware

V3.0.8.0 (06/17/2021)
==============================================
- Add SHA256 self-sign support for dpscat and inf-wizard2

- Update kBench usb ids

- Fix inf-wizard2 re-pack (Now includes 7za.exe)

- Add support for "DeviceInterfaceGUID" (singular)
```

mcuee, 2021-07-20:
Next release: 3.1.x series for bug fixes if any; 3.2.0.0 if there are major features.

Ying-Pin-Wu, 2022-05-04:
Could please provide MinGW .a library ?

mcuee, 2022-05-04:
MinGW is not supported. However you can link to the DLL directly.

mcuee, 2022-05-04:
Probably gendef can help if you really want to create the import library.
https://sourceforge.net/p/mingw-w64/wiki2/gendef/
https://sourceforge.net/p/mingw-w64/wiki2/Answer%20generation%20of%20DLL%20import%20library/

mcuee, 2022-05-04:
Build log for MSYS 2 MinGW 32bit compiler.

```
$ make -f GNUmakefile
[K] Running examples SOLUTION makefile..
make --no-print-directory --directory=./hot-plug-monitor  all
[K] Multi-lib support = No
[K] Running hot-plug-monitor PROJECT makefile..
windres -I. -I.. -I../../includes --target=pe-i386 hot-plug-monitor_rc.rc -o bin/x86/hot-plug-monitor_rc.o
gcc -m32 -I. -I.. -I../../includes -mconsole -mwin32 -s -O2 -c hot-plug-monitor.c -o bin/x86/hot-plug-monitor.o
gcc -m32 -I. -I.. -I../../includes -mconsole -mwin32 -s -O2 -c ../examples.c -o bin/x86/examples.o
../examples.c: In function 'Examples_GetTestDeviceEx':
../examples.c:171:32: warning: passing argument 2 of 'LstK_Count' from incompatible pointer type [-Wincompatible-pointer-types]
  171 |         LstK_Count(deviceList, &deviceCount);
      |                                ^~~~~~~~~~~~
      |                                |
      |                                ULONG * {aka long unsigned int *}
In file included from ../examples.h:10,
                 from ../examples.c:25:
../../includes/libusbk.h:4328:24: note: expected 'PUINT' {aka 'unsigned int *'} but argument is of type 'ULONG *' {aka 'long unsigned int *'}
 4328 |             _ref PUINT Count);
      |                  ~~~~~~^~~~~
../examples.c: In function 'Bench_Configure':
../examples.c:243:91: warning: passing argument 5 of 'DriverAPI->ControlTransfer' from incompatible pointer type [-Wincompatible-pointer-types]
  243 |                 success = DriverAPI->ControlTransfer(UsbHandle, Pkt, (PUCHAR)TestType, 1, &transferred, NULL);
      |                                                                                           ^~~~~~~~~~~~
      |                                                                                           |
      |                                                                                           DWORD * {aka long unsigned int *}
../examples.c:243:91: note: expected 'PUINT' {aka 'unsigned int *'} but argument is of type 'DWORD *' {aka 'long unsigned int *'}
../examples.c:245:85: warning: passing argument 5 of 'UsbK_ControlTransfer' from incompatible pointer type [-Wincompatible-pointer-types]
  245 |                 success = UsbK_ControlTransfer(UsbHandle, Pkt, (PUCHAR)TestType, 1, &transferred, NULL);
      |                                                                                     ^~~~~~~~~~~~
      |                                                                                     |
      |                                                                                     DWORD * {aka long unsigned int *}
In file included from ../examples.h:10,
                 from ../examples.c:25:
../../includes/libusbk.h:3105:27: note: expected 'PUINT' {aka 'unsigned int *'} but argument is of type 'DWORD *' {aka 'long unsigned int *'}
 3105 |             _outopt PUINT LengthTransferred,
      |                     ~~~~~~^~~~~~~~~~~~~~~~~
gcc -m32 -I. -I.. -I../../includes -mconsole -mwin32 -s -O2 -o ./bin/x86/hot-plug-monitor.exe bin/x86/hot-plug-monitor_rc.o bin/x86/hot-plug-monitor.o bin/x86/examples.o -L../../bin/dll/x86/ -Wl,--kill-at,--enable-stdcall-fixup -lkernel32 -luser32 -lusbK
make --no-print-directory --directory=./load-driver-api  all
[K] Multi-lib support = No
[K] Running load-driver-api PROJECT makefile..
windres -I. -I.. -I../../includes --target=pe-i386 load-driver-api_rc.rc -o bin/x86/load-driver-api_rc.o
gcc -m32 -I. -I.. -I../../includes -mconsole -mwin32 -s -O2 -c load-driver-api.c -o bin/x86/load-driver-api.o
gcc -m32 -I. -I.. -I../../includes -mconsole -mwin32 -s -O2 -c ../examples.c -o bin/x86/examples.o
../examples.c: In function 'Examples_GetTestDeviceEx':
../examples.c:171:32: warning: passing argument 2 of 'LstK_Count' from incompatible pointer type [-Wincompatible-pointer-types]
  171 |         LstK_Count(deviceList, &deviceCount);
      |                                ^~~~~~~~~~~~
      |                                |
      |                                ULONG * {aka long unsigned int *}
In file included from ../examples.h:10,
                 from ../examples.c:25:
../../includes/libusbk.h:4328:24: note: expected 'PUINT' {aka 'unsigned int *'} but argument is of type 'ULONG *' {aka 'long unsigned int *'}
 4328 |             _ref PUINT Count);
      |                  ~~~~~~^~~~~
../examples.c: In function 'Bench_Configure':
../examples.c:243:91: warning: passing argument 5 of 'DriverAPI->ControlTransfer' from incompatible pointer type [-Wincompatible-pointer-types]
  243 |                 success = DriverAPI->ControlTransfer(UsbHandle, Pkt, (PUCHAR)TestType, 1, &transferred, NULL);
      |                                                                                           ^~~~~~~~~~~~
      |                                                                                           |
      |                                                                                           DWORD * {aka long unsigned int *}
../examples.c:243:91: note: expected 'PUINT' {aka 'unsigned int *'} but argument is of type 'DWORD *' {aka 'long unsigned int *'}
../examples.c:245:85: warning: passing argument 5 of 'UsbK_ControlTransfer' from incompatible pointer type [-Wincompatible-pointer-types]
  245 |                 success = UsbK_ControlTransfer(UsbHandle, Pkt, (PUCHAR)TestType, 1, &transferred, NULL);
      |                                                                                     ^~~~~~~~~~~~
[... cut ...]
../../includes/libusbk.h:3105:27: note: expected 'PUINT' {aka 'unsigned int *'} but argument is of type 'DWORD *' {aka 'long unsigned int *'}
 3105 |             _outopt PUINT LengthTransferred,
      |                     ~~~~~~^~~~~~~~~~~~~~~~~
gcc -m32 -I. -I.. -I../../includes -mconsole -mwin32 -s -O2 -o ./bin/x86/xfer-control-loop.exe bin/x86/xfer-control-loop_rc.o bin/x86/xfer-control-loop.o bin/x86/examples.o -L../../bin/dll/x86/ -Wl,--kill-at,--enable-stdcall-fixup -lkernel32 -luser32 -lusbK
```

mcuee, 2022-05-04:
For MSYS 2 mingw64, the detection of the 64bit architecture does not work. So I have to use the following dirty patch.
```
$ diff examples-project.mkinc examples-project_org.mkinc
53c53
< #ifeq ("$(ARCH)","amd64")
---
> ifeq ("$(ARCH)","amd64")
58,64c58,64
< #endif
< #ifneq ("$(ARCH)","amd64")
< #override ARCH:=x86
< #ARCH_MSVC:=Win32
< #MACHINE:=-m32
< #RC_PE_TARGET:=--target=pe-i386
< #endif
---
> endif
> ifneq ("$(ARCH)","amd64")
> override ARCH:=x86
> ARCH_MSVC:=Win32
> MACHINE:=-m32
> RC_PE_TARGET:=--target=pe-i386
> endif

```

Ying-Pin-Wu, 2022-05-04:
> Probably gendef can help if you really want to create the import library. https://sourceforge.net/p/mingw-w64/wiki2/gendef/ https://sourceforge.net/p/mingw-w64/wiki2/Answer%20generation%20of%20DLL%20import%20library/
> 
> Here are the MSYS2 mingw64/mingw32 import library for you testing. [mingw_msys2.zip](https://github.com/mcuee/libusbk/files/8618509/mingw_msys2.zip)

Thank you for your help, I can link library successful
 
I was trying to generate .def via reimp , but it fail and show "invalid or corrupt import library".
following your step by using gendef, it work for me .

mcuee, 2023-02-22:
Close this one for now. I will create new issues if there are changes and the changes are significant enough to make a new release.

closed 2023-02-22 by mcuee