mcuee/libusbk #23: How to build libusbk

Thompson-NO3, 2021-07-20:
I notice that build using make.cmd needs doscmd.exe
but there's nothing explaining what is doscmd.exe and where it comes from and its source code
Sadly, this tool doesn't run under XP SP3, then how to setup an environment to build?

TravisRo, 2021-07-20:
Using the home rolled make system (make.cmd) is difficult to get setup.  doscmd is just something that I threw together to help with several tasks.  It should work with windows XP but I've attached the source code and a WXP release binary regardless.

Look over make.cfg. You'll need to tailor that for your build environment.  Once everything is completely setup properly, you can use can use: "make dist" to build everything. 

For a simpler test, use something like:
make libusbK.dll fre x86 WXP

Here are all the sources you can build:
libusbK.sys; libusbK.lib; kList.exe; kBench.exe; dpscat.exe; libusbK.dll

I suggest using Visual Studios if you can tolerate the VC runtime dependency.
 
[doscmd-all-1.2.2.19.zip](https://github.com/mcuee/libusbk/files/6848487/doscmd-all-1.2.2.19.zip)
[doscmd_wxp.zip](https://github.com/mcuee/libusbk/files/6848495/doscmd_wxp.zip)

Thompson-NO3, 2021-07-22:
There are two problems till now.
1. I noticed a line !DCMD! -rp d "!__SRCS_DIR!\\" d "!G_BUILD_OUTPUT_BASE_ABS_DIR!" in make.cmd in libusbk source code.
but there's no code related in your doscmd source code. So after my build, the result won't work.
2. You use temp dir to store some bat file, if my account name is Chinese, not all English, the function wcstombs in doscmd.c will not work, it returns -1. I found I should setlocale to solve that.

I have tried to build the driver through make.cmd because I found it will do all the work not just build driver. I don't prefer to use Visual Studio for it's too big, boot slowly.

Thompson-NO3, 2021-07-23:
I also want to know how can I join this project

mcuee, 2021-07-23:
@Thompson-NO3 What do you mean by "join the project"? Contributions are always welcome.

For example, you can contribute by creating pulling requests if you find some issues and then have the fixes. You can also contribute by just submitting issues like this one.

Thompson-NO3, 2021-07-23:
Using github for a long time, but just clone other's repo.
In this issue, I have found some issue related to doscmd.exe, I just want to modify it and upload the new code, or pls let Travis upload the working code.
I also want to write some doc about building for I just find it a bit hard to build the releases.
I know PR and open issue. I just want to become like you have the right to push code.
I don't familiar with github, don't know how to apply to join a project, becoming a collaborator.
The github doc just tells me a project's maintainer can invite someone to join. I just want to be invited.

mcuee, 2021-07-23:
Thanks for the offer to help. But you have to start somewhere before we can invite you as a collaborator. You can actually start to create pull request for the following.
1) Document updates on how to build libusbk are more than welcome.
2) As for DOSCMD, let me create a git repo and then you can create issues and pull request for DOSCMD.

mcuee, 2021-07-23:
The Wiki is here. You can suggest how to improve the wiki.
https://github.com/mcuee/libusbk/wiki

mcuee, 2021-07-23:
The documentation is here, you are also welcome to look into it.
http://libusbk.sourceforge.net/UsbK3/index.html

TravisRo, 2021-07-23:
I'll put the doscmd source code the repository so you can create pull requests for it.

mcuee, 2021-07-23:
@TravisRo So you will put the doscmd source code in this libusbk repo? That is even better. Initially I was thinking of using a separate repo.

TravisRo, 2021-07-23:
Yes. I will bury it down in the "build_tools" folder

mcuee, 2021-07-24:
@Thompson-NO3 
doscmd source code has been added here. You are welcome to create pull-request. Thanks.
https://github.com/mcuee/libusbk/tree/master/libusbK/build_tools

mcuee, 2021-08-05:
Close this for now.

closed 2021-08-05 by mcuee