mcuee/libusbk #28: Libusbk packet loss

lianghaoyuan, 2021-08-13:
When using libusbk to test the USB camera, it is found that the pipe packet is lost

lianghaoyuan, 2021-08-13:
hk.txt:13687:  10.3                          73680  IN     07 06 07 07  07 07 07 06  06 07 07 07  06 07 07 07  ................  649us     13666.1.0        2021/08/13  09:53:58.147  unknown
hk.txt:13688:  10.3                              4  IN     07 06 06 06                                         ....              582us     13667.1.0        2021/08/13  09:53:58.147  unknown
hk.txt:13689:  10.3                              4  IN     07 06 06 06                                         ....              561us     13668.1.0        2021/08/13  09:53:58.148  unknown
hk.txt:13690:  10.3                              4  IN     07 06 06 06                                         ....              569us     13669.1.0        2021/08/13  09:53:58.148  unknown
hk.txt:13691:  10.3                              4  IN     07 06 06 06                                         ....              538us     13670.1.0        2021/08/13  09:53:58.149  unknown
hk.txt:13692:  10.3                              4  IN     07 06 06 06                                         ....              558us     13671.1.0        2021/08/13  09:53:58.149  unknown
hk.txt:72363:  10.3                          73680  IN     08 09 08 08  08 08 09 09  09 09 07 09  08 09 08 08  ................  693us     72342.1.0        2021/08/13  10:01:03.371  unknown
hk.txt:72364:  10.3                              4  IN     0b 0b 0a 0a                                         ....              512us     72343.1.0        2021/08/13  10:01:03.372  unknown
hk.txt:72365:  10.3                              4  IN     0b 0b 0a 0a                                         ....              565us     72344.1.0        2021/08/13  10:01:03.373  unknown
hk.txt:72366:  10.3                              4  IN     0b 0b 0a 0a                                         ....              563us     72345.1.0        2021/08/13  10:01:03.373  unknown
hk.txt:74156:  10.3                          73680  IN     0a 0b 0a 0b  09 0a 0b 0a  0a 0a 0b 0a  0c 0a 0b 0b  ................  1.8ms     74135.1.0        2021/08/13  10:01:16.361  unknown
hk.txt:74157:  10.3                              4  IN     08 08 09 09                                         ....              577us     74136.1.0        2021/08/13  10:01:16.362  unknown
hk.txt:74158:  10.3                              4  IN     08 08 09 09                                         ....              607us     74137.1.0        2021/08/13  10:01:16.363  unknown
hk.txt:74159:  10.3                              4  IN     08 08 09 09                                         ....              569us     74138.1.0        2021/08/13  10:01:16.363  unknown
hk.txt:74160:  10.3                              4  IN     08 08 09 09                                         ....              614us     74139.1.0        2021/08/13  10:01:16.364  unknown
hk.txt:95564:  10.3                         168964  IN     07 06 07 07  07 06 06 07  06 07 06 07  07 07 07 07  ................  844us     95543.1.0        2021/08/13  10:03:51.479  unknown
hk.txt:95565:  10.3                              4  IN     06 06 06 06                                         ....              556us     95544.1.0        2021/08/13  10:03:51.480  unknown
hk.txt:95566:  10.3                              4  IN     06 06 06 06                                         ....              561us     95545.1.0        2021/08/13  10:03:51.480  unknown
hk.txt:95568:  10.3                              4  IN     06 06 06 06                                         ....              138us     95547.1.0        2021/08/13  10:03:51.481  unknown
hk.txt:95569:  10.3                              4  IN     06 06 06 06                                         ....              564us     95548.1.0        2021/08/13  10:03:51.482  unknown
hk.txt:95570:  10.3                              4  IN     06 06 06 06                                         ....              546us     95549.1.0        2021/08/13  10:03:51.482  unknown

lianghaoyuan, 2021-08-13:
73680  IN     07 06 07 07  07 07 07 06  06 07 07 07  06 07 07 07  ................  649us     13666.1.0  
     4  IN     07 06 06 06                                         ....              582us     13667.1.0  
     4  IN     07 06 06 06                                         ....              561us     13668.1.0  
     4  IN     07 06 06 06                                         ....              569us     13669.1.0  
     4  IN     07 06 06 06                                         ....              538us     13670.1.0  
     4  IN     07 06 06 06                                         ....              558us     13671.1.0  
 73680  IN     08 09 08 08  08 08 09 09  09 09 07 09  08 09 08 08  ................  693us     72342.1.0  
     4  IN     0b 0b 0a 0a                                         ....              512us     72343.1.0  
     4  IN     0b 0b 0a 0a                                         ....              565us     72344.1.0  
     4  IN     0b 0b 0a 0a                                         ....              563us     72345.1.0  
 73680  IN     0a 0b 0a 0b  09 0a 0b 0a  0a 0a 0b 0a  0c 0a 0b 0b  ................  1.8ms     74135.1.0  
     4  IN     08 08 09 09                                         ....              577us     74136.1.0  
     4  IN     08 08 09 09                                         ....              607us     74137.1.0  
     4  IN     08 08 09 09                                         ....              569us     74138.1.0  
     4  IN     08 08 09 09                                         ....              614us     74139.1.0  
168964  IN     07 06 07 07  07 06 06 07  06 07 06 07  07 07 07 07  ................  844us     95543.1.0  
     4  IN     06 06 06 06                                         ....              556us     95544.1.0  
     4  IN     06 06 06 06                                         ....              561us     95545.1.0  
     4  IN     06 06 06 06                                         ....              138us     95547.1.0  
     4  IN     06 06 06 06                                         ....              564us     95548.1.0  
     4  IN     06 06 06 06                                         ....              546us     95549.1.0

lianghaoyuan, 2021-08-13:
I need to process USB data very quickly, because libusbk does not process data fast enough, resulting in data loss

mcuee, 2021-08-13:
You have to provide more details: your devices, codes, what you are trying to achieve, etc.

mcuee, 2021-08-13:
For example, if this is a USB camera, does it use isochronous transfer or bulk transfer? That can be an important information. Your OS and test PC plays a part as well. The driver may play a part as well -- I found libusbk driver is less picky than WinUSB.

lianghaoyuan, 2021-08-13:
After the comparative test, it is found that it is really caused by the driver. There will be no problem in the official driver pressure test of USB camera. Batch transmission is used. The computer performance is very good. Can you help analyze it? It's been analyzed for two weeks and no reason has been found

lianghaoyuan, 2021-08-13:
The official driver is libusb-win32. I use libusbk. It is normal to test other USB cameras, such as Basler, IDS, Point Grey and so on. There is a problem with one camera. It is preliminarily judged that the driver processing is not fast enough or the data is lost due to program switching.

mcuee, 2021-08-13:
Please provide more details of how do you to "comparative test"? What is the official driver? You mentioned libusb-win32, that is supported by libusbk.dll. Then why do you want to use libusbk.sys?

mcuee, 2021-08-13:
Side topic, please change your github profile avatar. It is showing disrespect to the person.

closed 2021-08-13 by lianghaoyuan

mcuee, 2021-08-13:
No idea why you delete all the comments. Anyway, it is okay that you closed the issue. I will just lock this conversation.