mcuee/libusbk #38: libusbk and libusbdotnet

mcuee, 2022-05-16:
Ref: https://github.com/LibUsbDotNet/LibUsbDotNet/issues/158

Supposedly libusbk.sys is almost a drop in replacement of libusb0.sys, so by right it should work together with libusbdotnet. Somehow it does not. I am not so sure why.

travis012, 2022-06-24:
I haven't used LibUsbDotNet in quite a long time.  I will try and carve out some time to look at it.  The last time I used it, it actually worked for libusbK, WinUSB and it also had a mode that allowed it to use libusb-1.0.

mcuee, 2022-06-25:
@travis012 

Yes I was pretty sure it used to work as well. 

There is a pull request to fix this issue. Maybe you can take a look.
* https://github.com/LibUsbDotNet/LibUsbDotNet/pull/162

mcuee, 2023-02-22:
The fixes have been merged for the libusbdotnet v2 branch.

closed 2023-02-22 by mcuee