mcuee/libusbk #40: libusbk  for windows ARM?

Sourab1801, 2022-06-24:
Hello @mcuee ,  when you are going to launch libusbk for windows 10 arm 64 bit?

TravisRo, 2022-06-24:
I don't have any plans for a libusbK ARM driver but we might be able to get the libusbK library working so it could be used with the WinUSB driver (assuming there is one)

mcuee, 2022-06-25:
@TravisRo 
Yes there are WinUSB driver for Windows on ARM.

Ref: similar request for Zadig.
https://github.com/pbatard/libwdi/issues/155

Sourab1801, 2022-06-25:
Thamks for reply @TravisRo & @mcuee ,
              so WinUSB will work behalf of libusbk driver?
Actually i worked on windows 10 with libusbk driver support, Now i am using windows 10  on arm on raspberry pi 4 and i want to use libusbk driver but libusbk is not supported in WINDOWS 10 ARM, So as per your suggestion I have to use Winusb driver behalf of libusbk in windows arm os it will act like libusbk driver ?

mcuee, 2022-06-25:
If you use libusbk API or libusb-1.0 API, then both libusbk.sys and WinUSB driver are supported.

Hwoever, currently libusbk API (libusbk.dll) does not support Windows on ARM yet.

If you use libusb-1.0 API (libusb-1.0.dll), it has support for Windows on ARM already. But since Zadig (as well as libusbk-inf-wizard) does not work under Windows on ARM yet, you may need to change your device firmware to add WinUSB Compat ID (Microsoft OS Descriptor) so that WinUSB driver can be automatically installed.

mcuee, 2022-06-25:
BTW, this is the same as #35 
* #35

mcuee, 2023-02-08:
Here is the ARM64 VS2019 build for libusbK: the library (DLL and static lib), kBench, kList and dpscat.

Take note libusbK.sys as it is not meant for ARM/ARM64 platform. 

[libusbK_bin_arm64.zip](https://github.com/mcuee/libusbk/files/10680835/libusbK_bin_arm64.zip)

mcuee, 2023-02-08:
@Sourab1801 

Please give the above libusbK ARM64 binary a try to see if it works or not. Thanks.

mcuee, 2023-02-08:
@ArminiusTux

Please give the above libusbK ARM64 binary a try. You can try to run things like dpscat and kList. Thanks.

mcuee, 2023-09-24:
For those who are interested, please also refer to this PR.
* https://github.com/mcuee/libusbk/pull/52

jensenhuangyankai, 2024-08-30:
hi @mcuee, I'm a bit new to this project, but I'm interested in testing for ARM64. How can I install the binary you mentioned earlier?

mcuee, 2024-08-30:
> hi @mcuee, I'm a bit new to this project, but I'm interested in testing for ARM64. How can I install the binary you mentioned earlier?

Please try PR#52.
* https://github.com/mcuee/libusbk/pull/52