mcuee/libusbk #42: Bluescreen with loose HDMI connection

FrostKiwi, 2022-11-26:
A loose HDMI connector under vibration causes an Elgato CamLink to restart due to momentary loss of HDMI signal. When this happens during streaming via libUVC + libusbK from said connection, Windows Bluescreens and reboots. 100% repeatable.

https://user-images.githubusercontent.com/60887273/204079682-6c978ec1-d10b-4b24-a84c-e1c68f8483bd.mp4

Giving attribution is hard, but I suspect libusbK being involved, so I post it here. My exact setup surrounding this bug is posted in detail here https://github.com/libuvc/libuvc/issues/245. Windows 11, Version 21H2 build 22000.1219
[image]

mcuee, 2022-11-26:
Pretty sure this is not a libusbk problem but libusb problem. And there is no solution as now since libusb still does not support hotplug under Windows.

mcuee, 2022-11-26:
You can post the issue in libusb github along libusb debug. I will close the issue here.

closed 2022-11-26 by mcuee