mcuee/libusbk #46: Is there any way to overwrite Configuration Descriptor?

kusaanko, 2023-02-26:
Hi.

I have a usb device with wrong configuration descriptor.
I installed libusbk using Zadig 2.7, then it is recognized by LibUsbDotNet, but I can't communicate with the usb device (HID Device).

Here is the usb descriptor from Thethycon USB Descriptor Dumper.
```
------------------------------
Device Descriptor:
------------------------------
0x12	bLength
0x01	bDescriptorType
0x0110	bcdUSB
0xFF	bDeviceClass      (Vendor specific)
0x04	bDeviceSubClass   
0x00	bDeviceProtocol   
0x08	bMaxPacketSize0   (8 bytes)
0x0AE4	idVendor
0x0004	idProduct
0x0100	bcdDevice
0x01	iManufacturer
0x02	iProduct     
0x03	iSerialNumber
0x01	bNumConfigurations
Hex dump: 
0x12 0x01 0x10 0x01 0xFF 0x04 0x00 0x08 0xE4 0x0A 
0x04 0x00 0x00 0x01 0x01 0x02 0x03 0x01 

-------------------------
Configuration Descriptor:
-------------------------
0x09	bLength
0x02	bDescriptorType
0x0029	wTotalLength   (41 bytes)
*** ERROR: Found error in descriptor: incomplete descriptor
0x01	bNumInterfaces
*** ERROR: Found error in descriptor: incomplete descriptor
0x01	bConfigurationValue
0x00	iConfiguration
0xA0	bmAttributes   (Bus-powered Device, Remote-Wakeup)
0x0A	bMaxPower      (20 mA)
Hex dump: 
0x09 0x02 0x29 0x00 0x01 0x01 0x00 0xA0 0x0A 

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x00	bInterfaceNumber
0x00	bAlternateSetting
0x01	bNumEndPoints
*** ERROR: Found error in descriptor: incomplete descriptor
0x03	bInterfaceClass      (Human Interface Device Class)
0x00	bInterfaceSubClass   
0x00	bInterfaceProtocol   
0x00	iInterface
Hex dump: 
0x09 0x04 0x00 0x00 0x01 0x03 0x00 0x00 0x00 

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x81	bEndpointAddress  (IN endpoint 1)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0008	wMaxPacketSize    (1 x 8 bytes)
0x14	bInterval         (20 frames)
Hex dump: 
0x07 0x05 0x81 0x03 0x08 0x00 0x14 

*** ERROR: Specified length (0x73) of descriptor (type: 0x55) is greater than the remaining descriptor data 
Microsoft OS Descriptor is not available. Error code: 0x000003E3

--------------------------------
String Descriptor Table
--------------------------------
Index  LANGID  String
0x00   0x0000  
Hex dump: 

0x01   0x0000  Request failed with 0x000001B1
Hex dump: 

0x02   0x0000  Request failed with 0x000001B1
Hex dump: 

0x03   0x0000  Request failed with 0x000001B1
Hex dump: 

------------------------------

Whole Device Descriptor as hex dump: 
0x12, 0x01, 0x10, 0x01, 0xFF, 0x04, 0x00, 0x08, 0xE4, 0x0A, 
0x04, 0x00, 0x00, 0x01, 0x01, 0x02, 0x03, 0x01

Whole Configuration Descriptor as hex dump: 
0x09, 0x02, 0x29, 0x00, 0x01, 0x01, 0x00, 0xA0, 0x0A, 0x09, 
0x04, 0x00, 0x00, 0x01, 0x03, 0x00, 0x00, 0x00, 0x07, 0x05, 
0x81, 0x03, 0x08, 0x00, 0x14, 0x73, 0x55, 0x7E, 0x7B, 0xCA, 
0xD8, 0x01, 0x20, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
0x00

------------------------------
```

I found the configuration descriptor's wTotalLength 0x0029 is incorrect. 0x0019 is correct.
I forked libusbk and fix to override wTotalLength to 0x0019. Then the device worked!
But I don't have code certificate because it is so expensive, so I don't want to use the method.
So I want to override wTotalLength(or configuration descriptor) or communicate with the device with wrong configuration descriptor.
Is there any way to communicate with it?

mcuee, 2023-02-26:
A few questions.
1) Does Windows recognize the device as a USB HID device? You can check Device Manager.

If your device is not recongized by Windows, then you have to fix your device firmware. libusbK will not help in that case.

2) If it is reconigized as a USB HID device, then the configuration descriptor is probably not an issue any more. You should use HIDAPI to talk to your device and not libusbdotnet or libusbK.

Are you saying that you modify `libusbK.sys` to accomodate your device? That should not be the right approach.

kusaanko, 2023-02-26:
My device’s manufacturer bankrupted, so I can’t modify the firmware.
Windows can’t recognize the device. Device Manager shows “Unknown Device”
Then, you mean that I can’t use my device with libusbK?

mcuee, 2023-02-26:
> I forked libusbk and fix to override wTotalLength to 0x0019. Then the device worked!

Do you mean you modify libusbK.sys to work with your device? If yes that is a potential solution. But as you mentioned, you need to get a digital certificate to sign the driver.

mcuee, 2023-02-26:
> My device’s manufacturer bankrupted, so I can’t modify the firmware. Windows can’t recognize the device. Device Manager shows “Unknown Device” Then, you mean that I can’t use my device with libusbK?

I believe the standard solution is to write a filter driver to mask the problem of device descriptor, I believe "lower filter driver". 

But this is an advanced topic and you will still need to get a digital certificate (EV certificate) to sign the driver.

If you are interested in this solution, you can post the question to OSR NTDEV forum.
https://community.osr.com/categories/ntdev

kusaanko, 2023-02-26:
When I modified libusbK.sys to ignore device’s raw data, then the device worked with Windows Test mode.
But I don’t have certificate, so I can’t use this method.
I tried libusb-win32 too, and it doesn’t work.
If I can’t use vanilla libusbK(or other libusb), I will give up.

mcuee, 2023-02-26:
I see. In that case, I tend to think you have to give up. We can not help you here.

closed 2023-02-26 by mcuee