mcuee/libusbk #47: LIBUSB_MAX_CONTROL_TRANSFER_TIMEOUT avoid infinite timeout

fabriceo, 2023-04-28:
Hi team;
thanks for providing this driver to the community, it is especially useful for the device including DFU features.

I have realized in recent experimentation that the timeout parameter attached to libusb_control_transfer() cannot be infinite as there is a test in the source code which maximize it to the value of LIBUSB_MAX_CONTROL_TRANSFER_TIMEOUT, which is defined in sys/drv_private.h.

this is a problem because the DFU protocol erase the flash memory at the first control request and this can take more than 5 seconds ! 

would you be so kind to change the value to a larger one (say 30 seconds) and make a new release ? or to change the different tests against this value and maximize it to say 30seconds when the value given is 0, so that is closer to infinite :)

many thanks

mcuee, 2023-04-29:
Hmm, by right you do not need to change this in the driver, rather you can set your time out in your application.

```
// Used as a timeout override for some control request; most will use either the pipe policy timeout or if not set, the request.timeout.
// If neither are set the timeout is infinite.
#define LIBUSB_MAX_CONTROL_TRANSFER_TIMEOUT 5000
```

mcuee, 2023-04-29:
BTW, please refer to this issue as well. You can say there will be no further new release of libusbK.sys since we do not have an EV certificate for driver attestation signing.
* https://github.com/mcuee/libusbk/issues/26

mcuee, 2023-04-29:
If you find out that `libusb_control_transfer()` cannot be infinite when using libusbK.sys driver, then please raise an issue in libusb github (https://github.com/libusb/libusb/issues). I do not beleieve it is true though.

fabriceo, 2023-05-03:
Thank you for your comments.
you are right that the timeout problem I m referring is not related to libusbK driver.
After many tests on many OS (OSX 86, OSX arm64, rpi arm32, rpi arch64, ubuntu 22 x86_64, window 7 and windows 10) the libusb_control_transfer request doesn't timeout with the same behavior!
on windows 10, I can get an infinite timeout (with value 0) when compiling with mingw64 and adding static library libusb1.0.26. using winusb or libusbK gives same result.
but on windows 7 with the same executable and driver I get a 5 seconds timeout.

Infinite timeout also on for OSX with utility compiled with static library 1.0.26.
Same on rpi32 and rpi64 with dynamic library 1.023 and 1.024

But on Ubuntu 20.04 running in virtualbox I get the 5 seconds timeout with dynamic library libusb1.0.25 and 1.0.26 ...

so a problem exists but nothing to do with libusbK thus I m closing this "non" issue.
Thanks again

fabriceo, 2023-05-03:
this was a non issue for libusbK

mcuee, 2023-05-04:
If you think this is an issue with libusb, you may want to raise an issue with libusb project.

closed 2023-05-11 by mcuee