mcuee/libusbk #51: Driver installation failed on Mi 5 phone

5tiaowu, 2023-07-30:
[image]

mcuee, 2023-07-31:
You need to provide more info, like the Windows OS version you are using.

Is this using x86 or x64? ARM or ARM64 is not supported.

5tiaowu, 2023-07-31:
> You need to provide more info, like the Windows OS version you are using.
> 
> 
> 
> Is this using x86 or x64? 
> 
> 
> 
> Windows on ARM or ARM64 is not supported.

I am using the win10 x66 system. I want to know if this is a mobile device problem or a computer problem.

mcuee, 2023-07-31:
Please try Zadig to see if it works.

5tiaowu, 2023-07-31:
> Please try Zadig to see if it works.

Can't

[image]

5tiaowu, 2023-07-31:
> Please try Zadig to see if it works.
> https://zadig.akeo.ie/

Can't[image]

mcuee, 2023-07-31:
Then you have to use another computer.

5tiaowu, 2023-07-31:
> Then you have to use another computer.

Do you know what the problem is? Is it a system problem or a device problem?

mcuee, 2023-07-31:
> Do you know what the problem is? Is it a system problem or a device problem?

Should be a system problem.

5tiaowu, 2023-07-31:
> > Do you know what the problem is? Is it a system problem or a device problem?
> 
> 
> 
> Should be a system problem.
> 
> 

Isn't it all right as long as it is x86 or 64?

mcuee, 2023-07-31:
> Isn't it all right as long as it is x86 or 64?

You may want to raise an issue in libwdi project as the error code is new to me. Please post the full Zadig debug log there.
https://github.com/pbatard/libwdi/issues

mcuee, 2023-07-31:
As this is related to libwdi, I will close this issue. We can follow up in libwdi once you raise the issue.

mcuee, 2023-07-31:
You can still post the full Zadig debug info here.

closed 2023-07-31 by mcuee