mcuee/libusbk #54: BSOD when unplugging USB device during activity

sonatique, 2023-09-27:
While doing tests of removing USB device abruptly (i.e. physically upluggin it) during activity (basically reading on a data enpoint in a loop) I systematically got a BSOD.

My stack is an application using libusb-1.0, libubsK.dll and libuskK.sys (latest signed versions)

Analysis of the minidump using WinDbg gave the following output, in which libusbK.sys is clearly identified as doing something wrong.

It is a bit beyond my knowledge, could someone have a look at it? I guess it's easily reproducible.. Thanks

Here is WinDbg output:

```
*******************************************************************************
*                                                                             *
*                        Bugcheck Analysis                                    *
*                                                                             *
*******************************************************************************

KERNEL_SECURITY_CHECK_FAILURE (139)
A kernel component has corrupted a critical data structure.  The corruption
could potentially allow a malicious user to gain control of this machine.
Arguments:
Arg1: 0000000000000003, A LIST_ENTRY has been corrupted (i.e. double remove).
Arg2: fffffe80dfc26870, Address of the trap frame for the exception that caused the BugCheck
Arg3: fffffe80dfc267c8, Address of the exception record for the exception that caused the BugCheck
Arg4: 0000000000000000, Reserved

Debugging Details:
------------------

KEY_VALUES_STRING: 1

    Key  : Analysis.CPU.mSec
    Value: 1781

    Key  : Analysis.Elapsed.mSec
    Value: 11773

    Key  : Analysis.IO.Other.Mb
    Value: 1

    Key  : Analysis.IO.Read.Mb
    Value: 4

    Key  : Analysis.IO.Write.Mb
    Value: 12

    Key  : Analysis.Init.CPU.mSec
    Value: 109

    Key  : Analysis.Init.Elapsed.mSec
    Value: 60213

    Key  : Analysis.Memory.CommitPeak.Mb
    Value: 114

    Key  : Bugcheck.Code.LegacyAPI
    Value: 0x139

    Key  : Dump.Attributes.AsUlong
    Value: 1808

    Key  : Dump.Attributes.DiagDataWrittenToHeader
    Value: 1

    Key  : Dump.Attributes.ErrorCode
    Value: 0

    Key  : Dump.Attributes.KernelGeneratedTriageDump
    Value: 1

[... 206 lines cut ...]
fffffe80`dfc26ea0 ffff9304`e5492fa8     : ffff9304`d23f1020 00000000`00000010 00000000`00000000 00006cfb`239a7238 : 0x4000
fffffe80`dfc26ea8 ffff9304`d23f1020     : 00000000`00000010 00000000`00000000 00006cfb`239a7238 00006cfb`1ab6d488 : 0xffff9304`e5492fa8
fffffe80`dfc26eb0 00000000`00000010     : 00000000`00000000 00006cfb`239a7238 00006cfb`1ab6d488 00000000`00004000 : 0xffff9304`d23f1020
fffffe80`dfc26eb8 00000000`00000000     : 00006cfb`239a7238 00006cfb`1ab6d488 00000000`00004000 00000000`00004000 : 0x10

SYMBOL_NAME:  libusbK+33c1

MODULE_NAME: libusbK

IMAGE_NAME:  libusbK.sys

STACK_COMMAND:  .cxr; .ecxr ; kb

BUCKET_ID_FUNC_OFFSET:  33c1

FAILURE_BUCKET_ID:  0x139_3_CORRUPT_LIST_ENTRY_libusbK!unknown_function

OSPLATFORM_TYPE:  x64

OSNAME:  Windows 10

FAILURE_ID_HASH:  {3e2f28bf-7cbf-dc2c-f08f-3b94cf8f94c5}

Followup:     MachineOwner
```

mcuee, 2023-09-27:
Hmm, this is beyond my capability as well. I have not seen such issue before. 

@TravisRo 
Needs your help here.

VadimAspirin, 2024-03-11:
Hello,
Faced the same problem, has anyone found a solution?

TravisRo, 2024-03-12:
Can you switch the driver to winusb.sys?