mcuee/libusbk #58: VID 0000 PID 0006

0wwafa, 2025-01-01:
Is there a way to open a device with VID 0000 and PID 0006 with libusbk in windows?

On linux I can using libusb, but on windows it seems it's not possible, but it would be great to unbrick some devices.

mcuee, 2025-01-01:
Is the device recognized by Windows as a valid USB device? 000 is not a valid vendor ID.

0wwafa, 2025-01-15:
Yes, it is. it's a special case. A device has an eeprom and in the eeprom can be configure any vid/pid.
On Linux, having vid 0000 does not cause any problems with libusb, I can access the device, reprogram the eeprom and reboot it.
Instead on windows I could not do that.

mcuee, 2026-05-21:
This is probably due to the definition here. So I think we will not fix this any time.
https://github.com/mcuee/libusbk/blob/master/libusbK/includes/libusbk.h
```
 	//! USB-IF vendor ID
	USHORT idVendor;

	//! USB-IF product ID
	USHORT idProduct;
```